PikoRT/pikoRT #2: Need make twice to make thing work first time

mlouielu, 2017-08-09:
Found this issue

```
$ make
...
A    ../cmsis/TARGET_STM/pwmout_api.c
A    ../cmsis/TARGET_STM/rtc_api.c
A    ../cmsis/TARGET_STM/rtc_api_hal.h
A    ../cmsis/TARGET_STM/serial_api.c
A    ../cmsis/TARGET_STM/serial_api_hal.h
A    ../cmsis/TARGET_STM/sleep.c
A    ../cmsis/TARGET_STM/stm32_assert.h
A    ../cmsis/TARGET_STM/stm_spi_api.c
A    ../cmsis/TARGET_STM/trng_api.c
A    ../cmsis/TARGET_STM/us_ticker_16b.c
A    ../cmsis/TARGET_STM/us_ticker_32b.c
Exported revision 27126.
mkdir -p ../cmsis/util
wget https://raw.github.com/ARMmbed/mbed-os/master/platform/mbed_preprocessor.h -P ../cmsis/util
--2017-08-09 09:43:37--  https://raw.github.com/ARMmbed/mbed-os/master/platform/mbed_preprocessor.h
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving raw.github.com... 151.101.88.133
Connecting to raw.github.com|151.101.88.133|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://raw.githubusercontent.com/ARMmbed/mbed-os/master/platform/mbed_preprocessor.h [following]
--2017-08-09 09:43:38--  https://raw.githubusercontent.com/ARMmbed/mbed-os/master/platform/mbed_preprocessor.h
Resolving raw.githubusercontent.com... 151.101.88.133
Connecting to raw.githubusercontent.com|151.101.88.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1401 (1.4K) [text/plain]
Saving to: ‘../cmsis/util/mbed_preprocessor.h’

mbed_preprocessor.h                                         100%[========================================================================================================================================>]   1.37K  --.-KB/s    in 0s      

2017-08-09 09:43:39 (16.1 MB/s) - ‘../cmsis/util/mbed_preprocessor.h’ saved [1401/1401]

wget https://raw.github.com/ARMmbed/mbed-os/master/platform/mbed_assert.h -P ../cmsis/util
--2017-08-09 09:43:39--  https://raw.github.com/ARMmbed/mbed-os/master/platform/mbed_assert.h
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving raw.github.com... 151.101.88.133
Connecting to raw.github.com|151.101.88.133|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://raw.githubusercontent.com/ARMmbed/mbed-os/master/platform/mbed_assert.h [following]
--2017-08-09 09:43:40--  https://raw.githubusercontent.com/ARMmbed/mbed-os/master/platform/mbed_assert.h
Resolving raw.githubusercontent.com... 151.101.88.133
Connecting to raw.githubusercontent.com|151.101.88.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4002 (3.9K) [text/plain]
Saving to: ‘../cmsis/util/mbed_assert.h’

mbed_assert.h                                               100%[========================================================================================================================================>]   3.91K  --.-KB/s    in 0s      

2017-08-09 09:43:41 (24.4 MB/s) - ‘../cmsis/util/mbed_assert.h’ saved [4002/4002]

HOSTCC\tpiko.lds
CC\t../cmsis/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/TARGET_DISCO_F429ZI/system_clock.o
AS\tarch/v7m-entry.o
CC\tarch/v7m-faults.o
AS\tarch/v7m-head.o
AS\tarch/v7m-svcall.o
CC\tdrivers/char/mem.o
CC\tdrivers/char/random.o
CC\tdrivers/mtd/mtdchar.o
CC\tdrivers/mtd/mtdcore.o
CC\tdrivers/mtd/mtdram.o
CC\tdrivers/serial/serialchar.o
CC\tdrivers/serial/serialcore.o
CC\tdrivers/serial/stm32f429.o
CC\tdrivers/timer/systick.o
CC\tdrivers/timer/timercore.o
CC\tkernel/bitops.o
CC\tkernel/cbuf.o
CC\tkernel/cond.o
CC\tkernel/config.o
CC\tkernel/faults.o
CC\tkernel/fs/fs.o
CC\tkernel/fs/proc.o
CC\tkernel/fs/readdir.o
CC\tkernel/fs/romfs.o
CC\tkernel/fs/tmpfs.o
CC\tkernel/fs/vfs.o
CC\tkernel/irq.o
CC\tkernel/main.o
CC\tkernel/mm/mm.o
CC\tkernel/mm/page.o
CC\tkernel/mm/slab.o
CC\tkernel/mutex.o
CC\tkernel/printk.o
CC\tkernel/resource.o
CC\tkernel/sched-o1.o
CC\tkernel/sched-rr.o
CC\tkernel/sched.o
CC\tkernel/signal.o
CC\tkernel/softirq.o
kernel/softirq.c: In function 'open_softirq':
kernel/softirq.c:30:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
     if (nr >= NR_SOFTIRQS)
     ^~
kernel/softirq.c:33:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  softirq_vec[nr].action = action;
  ^~~~~~~~~~~
CC\tkernel/syscall.o
CC\tkernel/task.o
CC\tkernel/thread.o
CC\tkernel/time.o
CC\tlibc/fcntl.o
CC\tlibc/filesystem.o
CC\tlibc/piko/mman.o
CC\tlibc/piko/stubs.o
AS\tlibc/piko/syscalls.o
CC\tlibc/pthread.o
CC\tlibc/signal.o
CC\tlibc/stdio.o
CC\tlibc/stdlib.o
CC\tlibc/time.o
CC\tlibc/ucontext.o
CC\tlibc/unistd.o
CC\tlibc/utils.o
AS\tlibc/v7m-pthread.o
CC\ttarget/stm32f429/halt.o
CC\ttarget/stm32f429/init.o
CC\tuser/cat.o
CC\tuser/echo.o
CC\tuser/exit.o
CC\tuser/halt.o
CC\tuser/ls.o
CC\tuser/reboot.o
CC\tuser/sh.o
user/sh.c: In function 'exec_command':
user/sh.c:44:5: warning: this 'for' clause does not guard... [-Wmisleading-indentation]
     for (cmd = (struct shell_cmd *) &__shell_cmd_start__;
     ^~~
user/sh.c:52:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'for'
  write(fd, TERM_CMD_NOT_FOUND, sizeof(TERM_CMD_NOT_FOUND) - 1);
  ^~~~~
OBJCOPY\tkernel/fs/version.o
LD\tpiko.elf
target/stm32f429/init.o: In function `__printk_init':
/home/grd/kernel/pikoRT/target/stm32f429/init.c:49: undefined reference to `HAL_GPIO_Init'
/home/grd/kernel/pikoRT/target/stm32f429/init.c:78: undefined reference to `HAL_UART_Init'
collect2: error: ld returned 1 exit status
make: *** [Makefile:55: piko.elf] Error 1
$ make
..
CC\tlibc/stdlib.o
CC\tlibc/time.o
CC\tlibc/ucontext.o
CC\tlibc/unistd.o
CC\tlibc/utils.o
AS\tlibc/v7m-pthread.o
CC\ttarget/stm32f429/halt.o
CC\ttarget/stm32f429/init.o
CC\tuser/cat.o
CC\tuser/echo.o
CC\tuser/exit.o
CC\tuser/halt.o
CC\tuser/ls.o
CC\tuser/reboot.o
CC\tuser/sh.o
user/sh.c: In function 'exec_command':
user/sh.c:44:5: warning: this 'for' clause does not guard... [-Wmisleading-indentation]
     for (cmd = (struct shell_cmd *) &__shell_cmd_start__;
     ^~~
user/sh.c:52:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'for'
  write(fd, TERM_CMD_NOT_FOUND, sizeof(TERM_CMD_NOT_FOUND) - 1);
  ^~~~~
OBJCOPY\tkernel/fs/version.o
LD\tpiko.elf
OBJCOPY\tpiko.hex
OBJCOPY\tpiko.bin
$
```

yenWu, 2017-08-09:
Yup, that's a really a problem. Due to Makefile process, it assigns all variable at begin, and we need to fetch related lib from "mbed" repo. So we can't use `wildcard` to retrieve file name into C source variable.

target/stm32p103/build.mk:
```
25 CSRC    += $(wildcard ../cmsis/TARGET_STM/TARGET_STM32F1/device/*.c)
```

jserv, 2017-08-10:
It should be resolved now.

yenWu, 2017-08-11:
Sorry to say that it couldn't work on `stm32f429`.

The same reason but different design between `stm32p103` and `stm32f429`. Yet we don't apply STM HAL to `stm32p103`, so it doesn't happen on it. I'm already working on it.

In short, it's still not resolved.

commit 16c8ad57 (2017-08-16): mk: Specify necessary HAL src for building

After this change, we could only `make` once without `prebuild`.

Close #2
commit cabc6e33 (2017-08-17): mk: Specify necessary HAL src for building

After this change, we could only `make` once without `prebuild`.

Close #2
closed 2017-08-16 by yenWu