PikoRT/pikoRT #8: MacOS make prebuild lsb_release: command not found

mlouielu, 2017-08-09:
```
$ make prebuild
make: lsb_release: Command not found
svn export --force https://github.com/ARMmbed/mbed-os/trunk/cmsis/ ../cmsis/arm
A    ../cmsis/arm
A    ../cmsis/arm/TARGET_CORTEX_A
A    ../cmsis/arm/TARGET_CORTEX_A/TOOLCHAIN_ARM
A    ../cmsis/arm/TARGET_CORTEX_A/TOOLCHAIN_ARM/cmsis_armcc.h
A    ../cmsis/arm/TARGET_CORTEX_A/TOOLCHAIN_ARM/cmsis_armclang.h
A    ../cmsis/arm/TARGET_CORTEX_A/cmsis_compiler.h
A    ../cmsis/arm/TARGET_CORTEX_A/core_ca9.h
```

Not sure if this will affect.

jserv, 2017-08-11:
Instead of depending on `lsb_release`, we might have to detect QEMU version to specify correct arguments. @mlouielu, can you attempt to work on this approach?

mlouielu, 2017-08-14:
Will do, but I'm not sure if that argument will affect, taking test tomorrow