PikoRT/pikoRT #14: TTY abstraction layer

yenWu, 2017-08-15:
Hi folks,

This a feature I'm working on. In the current implementation, we use `printk` without buffering. That would cause a poor performance and a huge section with interrupt disable. However, that's one of the reasons we want to import TTY abstraction layer. TTY layer could provide not only performance improvement but also better abstraction for the various device, such as UART, BLE, USB ... . It's also an good target to measure softirq we impl.

By the way, this's feature would be released soon. Furthermore, folks **should not work on this feature** directly, because we already have a prototype and specified API we want.

mlouielu, 2017-08-15:
@yenWu Would you like to use pull request and make a WIP for this feature?

yenWu, 2017-08-15:
@tonghuix , hmm. It just a half-done patch, and I don't think it would be helpful for you. Moreover, could you please tell me why do you need that? If there's anything within my ability, I'll pleasure to help you.

@mlouielu , sure ! That seem quite useful for documentation. I'll try it.

tonghuix, 2017-08-15:
@yenWu Just helping you developing...

yenWu, 2017-08-15:
@tonghuix , thank for your interest and concern. I prefer to do it by myself, but there's truly lots of work to do. But It also welcomes to improve or extend it after I finished. I could provide a brief impl guideline.

1. tty_core: main tty layer provide interface to user app
2. n_tty: serialization mechanism and buffering
3. serial_core: serial abstraction layer
4. stm32-uart: hardware specified uart operations

However, we just need a simplified version and retrieve the critical characteristics to re-impl all the stuff. That's all I thought important features. Any suggestions or cooperative tactic?

yenWu, 2017-08-15:
@tonghuix , Sorry, I thought I use a misleading word, `prototype`, on this issue. This prototype doesn't work yet. I just want to clarify I already have a preliminary idea and some progress on it. However, it would be presented as a PR, then you could review that.

tonghuix, 2017-08-16:
Thank you very much! I think your proposal are full fill for the feature, waiting your code now.