PikoRT/pikoRT #17: shell: Enhance parser of minishell

yenWu, 2017-08-17:
* No checking postfix
```
$ lsXXX
     1 .
     1 ..
     2 dev
     3 proc
```

* Parser fault
```
$ ls /dev/

-------------------------------------------------------------
 #UsageFault

 r0: 4e478048    r1: ce5000f6    r2: 170000f2    r3: 17608149
 r4: 00000000    r5: 00000000    r6: 00000000    r7: 00000000
 r8: 00000000    r9: 00000000   r10: 00000000   r11: 00000000
r12: 01f0424a    sp: 00000391    lr: bf62ca02    pc: 008f4ff3

EXC_RETURN: fffffff1
      ufsr: 00000000
-------------------------------------------------------------
```
* Parser fault
```
$ ls dev

-------------------------------------------------------------
 #UsageFault

 r0: 4e478048    r1: ce5000f6    r2: 170000f2    r3: 17608149
 r4: 00000000    r5: 00000000    r6: 00000000    r7: 00000000
 r8: 00000000    r9: 00000000   r10: 00000000   r11: 00000000
r12: 01f0424a    sp: 00000391    lr: bf62ca02    pc: 008f4ff3

EXC_RETURN: fffffff1
      ufsr: 00000000
-------------------------------------------------------------
```

mlouielu, 2017-08-17:
This is not the problem about `minishell`, is `ls.c` close the `NULL` pointer at the end.

yenWu, 2017-08-17:
@mlouielu , probably not. It also happends on `cat`.
```
$ catcatcatcat /proc/version
Piko version 0.0.0 (uscc@uscc-Inspiron-7559) (armv7-m, cortex-m3) #Thu Aug 17 09:36:32 2017
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

mlouielu, 2017-08-17:
You can take a look via gdb, `ls dev` and `ls /dev/` isn't a parser fault, it fault inside the `ls.c`.

But your example about `cat` and `lsXXX' is a parser problem.

yenWu, 2017-08-17:
@mlouielu , you're right. I'll tweak descriptions in this issue.

PR #30 (2017-08-22): Fix minishell parsing command
Close #17.
closed 2017-08-22 by jserv