PikoRT/pikoRT #31: Syscall code not compatible to linux?

mlouielu, 2017-08-22:
http://elixir.free-electrons.com/linux/latest/source/arch/arm64/include/asm/unistd32.h

Current syscall list (`syscall.c`) code isn't compatible to linux.

jserv, 2017-08-22:
Being compatible with Linux is NOT the goal of Piko/RT. Instead, PSE51 (minimal real-time system profile) should be appropriately an analog from the perspective of design.

Reference: https://www.linuxjournal.com/article/4444?page=0,1

jserv, 2017-08-22:
I will close this issue after primitive design documents are landed though.

mlouielu, 2017-08-22:
@jserv According to the feature matrix, PSE51 didn't contain process, does that mean the PSE51 don't need to implement `fork` or `clone` system call?

mlouielu, 2017-08-22:
Ah, I found the description of PSE51:

```

4.3.1 Minimal Real-time System Profile IEEE Std 1003.13 PSE51

This profile is intended for embedded systems, with a single
multi-threaded process, no file system, no user and group support and
only selected options from IEEE Std 1003.1b-1993.

```

jserv, 2017-08-22:
We do need `vfork` and `clone` system calls to validate `lmbench` on Piko/RT.

mlouielu, 2017-08-22:
Since current `pikoRT` didn't have `process` concept, will the implement of `vfork` and `clone` be create another thread, or `process` will be implemented in `pikoRT` in the future?

jserv, 2017-08-29:
superseded by #33

closed 2017-08-29 by jserv