PikoRT/pikoRT #49: Test suites don't expose `failed` on cases of forced termination

yenWu, 2017-10-15:
I could provide a little trick to reproduce this issue. Start with `make PLAT=stm32p103 check` and consecutively press `x` followed by `ctrl + a`. You would find out there's no failure.

jserv, 2017-10-15:
In other words, test runner should manipulate UNIX signals properly. Terminating QEMU is another case, and we don't really be aware of the external actions.

yenWu, 2017-10-15:
In current runner implementation, it can handle SIGINT well and list test cases which don't execute yet.

In case of QEMU, It's necessary to terminate QEMU due to system halt. However, It would quite often occur on developing a new feature. Take it as a failure rather than pass is good for developers to fix it and also pretty make sense very.

jserv, 2017-10-15:
We can utilize semihosting to perform bidirectional verification between host and device(s).