PikoRT/pikoRT #50: "ls -l" cause hard fault

akioolin, 2017-10-18:
hello, pikoRT developer:

Thank you for create this wonder land.

the following is how to generate the fault which i meet.

after OS boot sequence complete, the prompt will be displayed.
type the below 'ls' operation, will cause system hang up.

step 1. type 'ls'
step 2. type 'ls -l'
step 3. type 'ls'

the system will show the message like below:

#UsageFault

r0: 00000000 r1: 00003e73 r2: 00000002 r3: 00000014
r4: 00000008 r5: 200011f8 r6: 00000000 r7: 00003e73
r8: 00000002 r9: 00003688 r10: 20000f9c r11: 00000000
r12: 0000002d sp: 20028f90 lr: 000034a5 pc: 00002d82

EXC_RETURN: fffffff1
ufsr: 00000002 <INVSTATE>
-------------------------------------------------------------

-------------------------------------------------------------
#HardFault

r0: 00000000 r1: 00003e73 r2: 00000002 r3: 00000014
r4: 00000008 r5: 200011f8 r6: 00000000 r7: 00003e73
r8: 00000002 r9: 00003688 r10: 20000f9c r11: 00000000
r12: 0000002d sp: 20028f90 lr: 000034a5 pc: 00002d82

EXC_RETURN: fffffff1

for trying to understand the 'ls' command's flow, i try to add come code in the
below code piece.

    if (argc == 1)
        dir = opendir("/");  // FIXME: get current directory
    else {
        dir = opendir(argv[1]);
        printk("argv[1]=>%s, open %s\n", argv[1], (dir==NULL) ? "no dir":" done");
    }

    do {
        readdir_r(dir, &dirent, &result);
        if (result != NULL)
            printk("% 6d %s\n", dirent.d_ino, dirent.d_name);
    } while (result != NULL);
    closedir(dir);

the question is after step 3., the system will hang on. what is the root cause of this problem?
how to fix this problem ?

Thank you very much.

BR,
Akio

yenWu, 2017-10-19:
Thank for your report. I confirmed that it's a bug.

By the way, I revised the subject of this issue. Hard fault occurred in this situation results from that system is unable to handle usage fault anymore.

akioolin, 2017-10-19:
Thank you very much.

Is this easy to solve?

if I want to add new command, micropython porting could be a good starting point or not?

yenWu, 2017-10-20:
@akioolin , I answer this question, "Is this easy to solve", first. It might be easy to just fix this bug, but however, it's better to refactor shell command parser first.

Micropython porting is worked in place by @mlouielu. Please see #16 .

manbing, 2017-12-17:
Dear, piko-RT developer:
Excuse me, i am novice about ARM and OS.
May i ask some questions about this issue, Thank you very much.

For now piko-RT. this issue will output the below information
`
$ ls
     1 .
     1 ..
     2 dev
     3 proc
$ ls -l
$ ls
     1 .
     1 ..
     2 dev
     3 proc

-------------------------------------------------------------
 #UsageFault

 r0: 00000000    r1: 00003e52    r2: 00000002    r3: 00000014
 r4: 00000008    r5: 2000122c    r6: 00000000    r7: 00000002
 r8: 00003e52    r9: 00000000   r10: 00000000   r11: 00000000
r12: 00000000    sp: 20008f98    lr: 00003323    pc: 00002ce2

EXC_RETURN: fffffff1
      ufsr: 00000000
-------------------------------------------------------------
`
i know this issue is ls.c can't parse  "ls -l".
but, i have the below problems.
1.  why the usage fault exception is trigger on third command "ls", not second command "ls -l"? 
2. i do not sure whether the ufsr has value when usage fault is triggering?
    In this case, ufsr does not have value. how Developer know the root cause?